eclipse-threadx/netxduo #261: http(s) client: how to download a big file chunk by chunk ?

joelguittet, 2024-03-20:
Hello,

I'm trying to do some firmware update with a custom client application.
I successfully done several GET/PUT/POST operation with the https client on my device (stm32l4) with wifi offloading using the b-l4s5i-iot01a board.

Now I try to download a firmware file to perform the FOTA operation on my device. The file is currently about 650kB.
The skeleton/configuration I have used until now seems not working at all and totally blocking in the `nx_web_http_client_response_body_get` call.... :-(

Can you indicate a skeleton/working configuration to perform GET HTTP to retrieve a big file chunk by chunk ? I need to flash the content received block by block because of course the whole file retrieved from the server will not fit in the RAM!!

Is `nx_web_http_client_response_body_get` to be used ? Is it possible to have a callback invoked for each chunk of the received file ? I have seen topics about the `NX_DISABLE_PACKET_CHAIN` configuration, but not explored for the moment because the mqtt addon doesn't build disabling the packet chain feature...

Thanks for pointer/help/skeleton/code example that may help on this topic!

Joel

Alec-Davis-NZ, 2024-04-03:
I had similar issues (but not chunked) see [https://github.com/eclipse-threadx/netxduo/issues/177] 

On the STM32 forums [https://community.st.com/t5/stm32cubeide-mcus/azurertos-netxduo-unable-to-download-16k-or-larger-image-using/m-p/618241] i uploaded a simple project to download various sizes from httpbin.org, where you can specify the size of bin file

Still unresolved.

reza-hdd, 2024-06-14:
Can you please let me know if you managed the to resolve this issue. I'm also running in this issue now.

joelguittet, 2024-06-17:
Hello @reza-hdd 

I'm currently using the following skeleton, working for the moment, but should be cleanup and probably a better management of errors should be done (particularly, I'm not 100% sure of the treatments to do for different return values of `nx_web_http_client_response_body_get`):

```
while (NX_WEB_HTTP_GET_DONE != err /*content_length > 0*/) { //TODO better

        err = nx_web_http_client_response_body_get(&client, &recv_packet, NX_WAIT_FOREVER);

        if ((NX_SUCCESS == err) || (NX_WEB_HTTP_GET_DONE == err)) {
            /* Read HTTP status code */
            if (0 == *status) {
                *status = (int)strtol(recv_packet->nx_packet_data_start + 9, NULL, 10);
            }
            /* Transmit data received to the upper layer */
            if (recv_packet->nx_packet_length > 0) {
                NX_PACKET *tmp = recv_packet;
                while (NULL != tmp) {
                    if (0
                        != (ret = callback(MY_HTTP_EVENT_DATA_RECEIVED,
                                           tmp->nx_packet_prepend_ptr,
                                           (size_t)(tmp->nx_packet_append_ptr - tmp->nx_packet_prepend_ptr),
                                           params))) {
                        nx_packet_release(recv_packet);
                        goto END;
                    }
                    tmp = tmp->nx_packet_next;
                }
            }
            //        content_length -= read_length;
            nx_packet_release(recv_packet);
        } else {
            ret = -1;
            goto END;
        }
    }
```