eclipse-threadx/netxduo #273: Is Netx DUO IPv6 standard certified or compliant for an end-node?

shahab5, 2024-05-08:
Hi,

We're using the Netx Duo stack for IPv4 and IPv6 offerings and we want to confirm that the IPv6 implementation (for an end-node) is compliant and certified to the IPv6 standard. We understand that NetX Duo doesn't offer router functionality and implements an IPv6 end-node functionality. 

If it conforms, can you share any tool or test harness used for compliance testing?

Thanks,
Shahab

yuxinzhou5, 2024-05-22:
NetX Duo 5.5 passed the "IPv6 Ready Logo" certification program in 2010.  
https://www.ipv6ready.org/db/index.php/public/logo/02-C-000454/

yuxinzhou5, 2025-01-30:
Closing this issue as the answer was provided and no further actions required.

closed 2025-01-30 by yuxinzhou5