eclipse-threadx/netxduo #278: nx_packet_allocated return false for RISCV-64 bits arch

Alexbruliau, 2024-06-26:
Hello, 

I am encountering an issue when I used nx_packet_allocate because ULONG is considered to be 8 bytes for the RISCV-64 bits architecture. 
It worked perfectly on my previous target CORTEX-R4F.

Any idea ? 

[image]

Adrian-Skerratt, 2024-07-11:
In ThreadX, ULONG (and LONG) should be 32 bits. On other 64-bit processors ULONG is a defined to unsigned int, but not for RISCV64.

This is a bug in ThreadX, but I don't have access to a working RISCV-64 system to check that making this change doesn't cause any other issues.

Are you able to test this?

You need to change the definition of LONG and ULONG in tx_port.h for the riscv-64 port

https://github.com/eclipse-threadx/threadx/blob/485a02faec6edccef14812ddce6844af1d7d2eef/ports/risc-v64/gnu/inc/tx_port.h#L103

to match the correct implementation for a 64-bit processor:

https://github.com/eclipse-threadx/threadx/blob/485a02faec6edccef14812ddce6844af1d7d2eef/ports/cortex_a77/gnu/inc/tx_port.h#L86

fdesbiens, 2026-09-28:
@Alexbruliau, @EAlexJ, and @cpdpls, thank you for the investigation and hardware feedback. I’m sorry it took us so long to close the loop. The RV64 ThreadX port now defines `ULONG` as 32 bits and uses a 64-bit type for pointer alignment. That change was merged into ThreadX `dev` in eclipse-threadx/threadx#534 and addresses the width mismatch reported here. I’m closing this issue as fixed. If packet allocation still fails with current ThreadX and NetX Duo revisions, please open a new issue with the revisions, compiler, and a minimal reproducer.

closed 2026-09-28 by fdesbiens