eclipse-threadx/netxduo #280: NetX Secure error NX_SECURE_PKCS1_INVALID_PRIVATE_KEY

robert-kau, 2024-07-15:
Hi,

I'm developing an HTTP web server with TLS. I started using a certificate and private key from a NetX Duo example, and it works. Now, I'm trying to use my own self-signed certificate with an RSA private key. However, when I call the nx_secure_x509_certificate_initialize function, which subsequently calls other functions, eventually invoking _nx_secure_x509_pkcs1_rsa_private_key_parse, a specific condition is satisfied on line 148 of nx_secure_x509_pkcs1_rsa_private_key_parse.c and my application don't work:
`if (tlv_type != NX_SECURE_ASN_TAG_INTEGER || tlv_type_class != NX_SECURE_ASN_TAG_CLASS_UNIVERSAL)
{
    return(NX_SECURE_PKCS1_INVALID_PRIVATE_KEY);
}`
This is because tlv_type is different from NX_SECURE_ASN_TAG_INTEGER. I generated the private key using OpenSSL with this command: openssl genrsa -traditional -out private_key.key 2048.
Is this correct? **What is the correct way to generate a private key for use with TLS?**

Thanks for any help.

Adrian-Skerratt, 2024-07-23:
I tried to reproduce your issue but I didn't get any errors

These are the steps I took to generate a private key in the correct format:

- generated a private_key.key file using the openssl command given above
- copied private_key.key to private_key.stripped and edited that file to remove the first and last lines (the ones that say "-----BEGIN RSA PRIVATE KEY-----")
- Converted that file from base64 to binary using the command "base64 --decode private_key.stripped >private_key.bin"
- used xxd to convert the binary file to a c array

robert-kau, 2024-07-30:
Thanks for the info. I was using OpenSSL version 3.0, which generates the RSA key with a different and higher version than PKCS#1. I used the -traditional flag, including the conversion from PEM to DER, and it started working.

I'm just noticing a 5s delay in loading the http page with TLS, when compared to the connection without TLS. I'm investigating how to improve this.

Also, I'm trying to use an elliptic curve key, to use TLS 1.3.

Adrian-Skerratt, 2024-08-05:
For TLS 1.3, NetX Secure doesn't support RSA based certificates, only elliptic curve. 

If I remember correctly private key computation is quite processor intensive, so 5 seconds seems understandable on a micro-controller. The NetX Duo documentation does mention that you can offload cryptographic functions to hardware cryptographic peripherals on your micro-controller if available, but I can't find any examples of this in use.

If you think this issue is now fixed, could you please close it