eclipse-threadx/netxduo #294: Using NetXDuo on Linux for testing - examples seg fault?

vrzh, 2024-12-18:
Hi,

Sorry if this is not the correct forum for this question.

A similar question to #155, I would like to use NetXDuo on a Linux system for quick testing. It looks like the information was provided in an answer to #155 but unfortunately the link referenced there [https://learn.microsoft.com/en-us/training/modules/build-a-udp-echo-client/3-building-and-running-the-project](url) does not seem to point to the correct place any more.

Does anyone have an archive of the page, or can point me in the right direction please?

I have compiled ThreadX under Linux and can successfully run some of the demo programs, but for NetXDuo everything complies without issue, but for example `demo_netx_duo_tcp.c` just seg faults. Wanted to ask if this is still possible before I dig any deeper. I can provide more info if required.

fdesbiens, 2025-01-28:
Hi @vrzh. My apologies. I completely forgot to reply before the holidays. 

I was hoping to find a copy of the page in the Internet Archive but came back empty-handed.

Do you still need assistance? I can nudge the team to see if someone can provide guidance.

vrzh, 2025-02-02:
Yes, it would be amazing to find a copy of the information. I am still very much interested.