eclipse-threadx/netxduo #298: Content-Type : text/event-stream always sets the Connection to Close

mlozarita, 2025-01-27:
**Describe the bug**
A quick introduction first. I am implementing a REST-API running in STM32 as my target server. I can easily send to the server with POST/GET request and server sending the response with NO problem.

The problem is that when my server responds with a header Content-Type: text/event-stream, the connection got closed and client does not receive any events from the server.

I am using Content-Type: text/event-stream as one of my header parameters because I want to implement Server-Sent Event process (SSE) in my system.

Please also mention any information which could help others to understand
the problem you're facing:
- What target device are you using? 
   STM32H753ZITx.
- Which version of Eclipse ThreadX? 
   ver . 6.4.1
- What toolchain and environment? 
   STMCubeIDE postman as a client.

- What have you tried to diagnose or workaround this issue?

  I have a work around to this issue to make it work. I have modified 2 files.
 nx_web_http_server.c and nx_tcpserver.c.

In nx_web_http_server.c, function _nx_web_http_server_generate_response_header(),  
#ifndef NX_WEB_HTTP_SERVER_OMIT_CONTENT_LENGTH
    /* Determine if there is content_length.  */
    if(content_length || (**status_code_not_modified == NX_FALSE**))  -->>> Change this from NX_TRUEto NX_FALSE, this is to force the connection to be alive.
    {
#ifndef NX_WEB_HTTP_KEEPALIVE_DISABLE

        /* Place the "Connection" field in the header.  */
        if(server_ptr -> nx_web_http_server_keepalive == NX_TRUE)
        {

            /* Keepalive. */
            status += nx_packet_data_append(packet_ptr, "Connection: keep-alive", 22,
                    server_ptr -> nx_web_http_server_packet_pool_ptr, NX_WAIT_FOREVER);
        }
        else
#endif /* NX_WEB_HTTP_KEEPALIVE_DISABLE */
        {

        /* Close. */
        status += nx_packet_data_append(packet_ptr, "Connection: Close", 17,
                                        server_ptr -> nx_web_http_server_packet_pool_ptr, NX_WAIT_FOREVER);
        }

nx_tcpserver.c., function _nx_tcpserver_timeout_process(), see the last if statement below.

    /* Initialize TCP sockets. */
    for(i = 0; i < server_ptr -> nx_tcpserver_sessions_count; i++)
    {
        session_ptr = &server_ptr -> nx_tcpserver_sessions[i];

        /* Skip socket that is not used. */
        if(session_ptr -> nx_tcp_session_socket.nx_tcp_socket_state == NX_TCP_CLOSED)
        {
            continue;
        }

        /* Skip socket that is already timeout. */
        if(session_ptr -> nx_tcp_session_expiration == 0)
        {
            continue;
        }

       /* Add checking below to skip the timeout when keepalive is enabled. This is to prevent the connection to be disconnected.*/
        **if (session_ptr -> nx_tcp_session_tls_session.nx_secure_tls_tcp_socket->nx_tcp_socket_keepalive_enabled)
        {
        	continue;
        }**

**To Reproduce**
Steps to reproduce the behavior:
1. Build my project in STMCubeIDE.
2. Implement REST-API using NetXduo libraries.
3.  Using postman (as a client), send a POST request to the server with the following http header.
      Content-Type : text/event-stream
      Connection : keep-alive
4. When the server receives the request command from client that client wants to receive events from the server, then the server will respond with the following http header. 
    _nx_web_http_server_generate_response_header.
     Content-Type : text/event-stream
      Connection : keep-alive
     Cache-Control : no-cache

      After sending the above http header to postman, the callback function will return NX_WEB_HTTP_CALLBACK_COMPLETED.
 
5. Then in the server, there is a thread that throws a sample event to the postman using the function _nxe_web_http_server_callback_data_send().

**Expected behavior**
Connection between client and server should always be alive (until it is closed by either the client or server) and when the server sends an event to the client, the client can receive that event.

**Impact**
We want to use SSE instead of polling mechanism in our project, thus it is kind of a showstopper.

**Logs and console output**

**Additional context**
Add any other context about the problem here.

hwmaier, 2025-01-28:
We did implement SSE and did not have to modify any NetXDuo source code. The problem with keep-alive is that you would need a content length if you use the build in server responses.

You can create your own server response and use that, refer to example below:

```
    UINT sendSseResponse(NX_WEB_HTTP_SERVER *serverPtr, UINT requestType)
    {
        constexpr char SSE_RESPONSE_HEADER[] =
            NX_WEB_HTTP_VERSION " 200 OK\r\n"
            "Content-Type: text/event-stream\r\n"
            "Cache-Control: no-cache\r\n"
            "Connection: keep-alive\r\n"
            "\r\n";
        UINT rc;
        NX_PACKET *responsePacketPtr;

        if (requestType != NX_WEB_HTTP_SERVER_GET_REQUEST)
            return sendResponse(serverPtr, NX_WEB_HTTP_STATUS_METHOD_NOT_ALLOWED);

        SessionState::getCurrentSession(serverPtr)->setEventSource();

        rc = nx_web_http_server_response_packet_allocate(serverPtr, &responsePacketPtr, NX_WAIT_FOREVER);
        if (rc != NX_SUCCESS)
            return rc;

        rc = nx_packet_data_append(responsePacketPtr, (void *)SSE_RESPONSE_HEADER, strlen(SSE_RESPONSE_HEADER), getPool(), NX_WAIT_FOREVER);
        if (rc != NX_SUCCESS)
        {
            nx_packet_release(responsePacketPtr);
            return rc;
        }

        rc = nx_web_http_server_callback_packet_send(serverPtr, responsePacketPtr);
        if (rc != NX_SUCCESS)
        {
            nx_packet_release(responsePacketPtr);
            return rc;
        }

        // This type of request is kept alive if client has requested so.
        return NX_WEB_HTTP_CALLBACK_COMPLETED; // Success!
    }
```

mlozarita, 2025-01-28:
Hi Henrik,

Thank you! Appreciate your suggestion here. Yes, I thought I could just create my own server response as well because of that content length but that did not help, and I think that was the reason I have another modification in the nx_tcpserver.c as mentioned above.  

Anyway, I followed your code above and tested it, but the connection got closed straight away. See for my test results below.

[image]

[image]

With your code + my modification with nx_tcpserver.c as mentioned above, then it works again.
[image]

Am I missing something?

Thank you!

hwmaier, 2025-01-28:
If nx_web_http_server_keepalive is set to true server should not disconnect. The SSE client has to open the connection with the `keep-alive` header set, that's important. Check with Wireshark that your headers from client and server are correctly set and also check who is closing the connection.

mlozarita, 2025-01-28:
Just to add note on this, I can confirm that the nx_web_http_server_keepalive is set to true when function _nx_web_http_server_receive_data() is called.

When I run a debug on the nx_tcpserver.c, I noticed that the nx_tcp_session_expiration got expired in the _nx_tcpserver_timeout_process() which is called in the _nx_tcpserver_thread_entry(), thus **closing the connection**.
Which leads me to believe that it is the server that is closing the connection. Nonetheless, I can confirm this with wireshark.

hwmaier, 2025-01-28:
Ah yes, in our SSE dispatchEvent method which writes the SSE data, we have this line:
```
            sessionPtr->nx_tcp_session_expiration = serverPtr->nx_web_http_server_tcpserver.nx_tcpserver_timeout;
```
which prevents the timeout as it resets the time-out counter everytime we write data.

hwmaier, 2025-01-28:
I would agree that the SSE support could be better, but instead of changing the time-out code as suggested earlier I rather would add proper methods for sending SSE responses which then set the correct keep-alive headers and reset the timeout counters internally. That then also would not break any existing behaviour.

mlozarita, 2025-01-28:
I think it should be independent from SSE response. What if the next event will happen after some time (ex. in 2 hours), then the connection will be closed again unless if we put that resetting of timeout counters somewhere in a thread that can reset it quickly before it kicks the timeout and disconnect it.

hwmaier, 2025-01-28:
There is no harm closing the connection if the events are 2 hours apart. The reason for a keep alive is to minimise connection/disconnection overheads for successive transmissions. It the transmissions are 2 h apart then the overhead of a few milliseconds to re-open the connection does not really matter. The client would reopen it anyway when it want to send a new event.

hwmaier, 2025-01-28:
You could override the __nx_web_http_server_connection_timeout function with your own implementation and set `nx_tcpserver_connection_timeout` to it. Your application specific timeout handler can then check if the session relates to SSE and not disconnect.

fdesbiens, 2025-01-28:
Thank you for submitting this issue, @mlozarita. And thank you for jumping in, @hwmaier! We appreciate the constructive feedback you provided.  

@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors:  Anything to add to this conversation?

@mlozarita Do you still feel a code change is required?

mlozarita, 2025-01-29:
Hi @fdesbiens, I believe it would be great if this can be fixed in the library layer rather than in the application layer.

Thank you!

intector, 2025-04-11:
Hey Guys,

I've been fighting the same issue that @mlozarita encountered. Last night, I modified the firmware, and now it closes the connection after the first message.

Here are my changes:
`
#ifndef NX_WEB_HTTP_SERVER_OMIT_CONTENT_LENGTH
    // ********************************************************************************************************
    // don't force "Connection: Close" if the content type is "text/event-stream" 
    // ********************************************************************************************************

    // Determine if there is content_length or content type is "text/event-stream"
    if (content_length || (status_code_not_modified == NX_TRUE) || (strcmp(content_type, "text/event-stream") == 0))
    {
#ifndef NX_WEB_HTTP_KEEPALIVE_DISABLE

        /* Place the "Connection" field in the header.  */
        if(server_ptr -> nx_web_http_server_keepalive == NX_TRUE)
        {

            /* Keepalive. */
            status += nx_packet_data_append(packet_ptr, "Connection: keep-alive", 22,
                    server_ptr -> nx_web_http_server_packet_pool_ptr, NX_WAIT_FOREVER);
        }
        else
#endif /* NX_WEB_HTTP_KEEPALIVE_DISABLE */
        {

        /* Close. */
        status += nx_packet_data_append(packet_ptr, "Connection: Close", 17,
                                        server_ptr -> nx_web_http_server_packet_pool_ptr, NX_WAIT_FOREVER);
        }

        /* Place the <cr,lf> into the buffer.  */
        status += nx_packet_data_append(packet_ptr, crlf, 2, server_ptr -> nx_web_http_server_packet_pool_ptr, NX_WAIT_FOREVER);

`
I added the check for the content_type, preventing skipping content_length = 0. The **_server_ptr->nx_web_http_server_keepalive_** must be set to **_TRUE_** before the **_nx_web_http_server_callback_generate_response_header_** call like so:

`
    server_ptr->nx_web_http_server_keepalive = NX_TRUE;

    // Build a response header with server status OK and additional CORS header
    status = nx_web_http_server_callback_generate_response_header(server_ptr,
                                                                  &sse_packet_ptr,
                                                                  NX_WEB_HTTP_STATUS_OK,
                                                                  0, // Set content length to 0 for SSE
                                                                  "text/event-stream",
                                                                  "Cache-Control: no-cache\r\n"
                                                                  "Connection: keep-alive\r\n"
                                                                  "Accept: text / event - stream\r\n"
                                                                  "Access-Control-Allow-Origin: *\r\n");

`

Who in their right mind would assume that defining NX_WEB_HTTP_SERVER_OMIT_CONTENT_LENGTH would solve this issue?
However, the connection is still closed after the first message, but I assume that is an issue in the nx_tcpserver.c. 

Today, I was searching the internet and stumbled upon this post. I already thought I was the only person on this planet trying to send SSE from an STM32H7. I'll try some of your ideas and see if it gets better.

Thank you, and to everyone else out there:

**_The secret is to keep banging the rocks together, guys._**

cypherbridge, 2026-02-09:
It looks like nx_tcpserver.c., function _nx_tcpserver_timeout_process() should probably be updated since any use case to persist TCP connection will need this.

About the SSE web application perhaps the application developers can contribute updates for review.

fdesbiens, 2026-09-28:
A proposed fix is available for review in #450, with documentation in eclipse-threadx/rtos-docs-asciidoc#83. It adds APIs for sending server-sent events and keeps event streams open while idle. Feedback from anyone affected by this issue would be welcome.