eclipse-threadx/netxduo #311: Unable to download a big file over HTTPS

joelguittet, 2025-04-02:
Hello,

I'm creating an application using NX Web HTTP client + NX Secure (HTTPS), Threadx and NetXDuo are v6.4.1. This works properly with some small GET/POST requests so far. Now I would like to perform firmware update using this HTTPS client, file size is about 1MB.

The board is B-L4S5I-IOT01A, and so I'm using an offloading driver to interface with the Wi-Fi module. The driver is a bit modified from https://github.com/eclipse-threadx/getting-started/tree/master/STMicroelectronics/B-L4S5I-IOT01A but mostly the same. It has a thread to receive data from the server, which priority is `nx_ip priority + 1`. The code snippet of the initialization follows:
```
tx_thread_info_get(tx_thread_identify(), NX_NULL, NX_NULL, NX_NULL, &priority, NX_NULL, NX_NULL, NX_NULL, NX_NULL);
tx_thread_create(&nx_driver_thread, "nx_driver_thread", nx_driver_thread_entry, 0, nx_driver_thread_stack, NX_DRIVER_STACK_SIZE, priority + 1, priority + 1, TX_NO_TIME_SLICE, TX_DONT_START)))
```

The `nx_ip` itself is created as shown, the `nx_ip` priority is 16, the ARP cache size is 1024 (also tried 512), and the `nx_ip_pool` size is `((1536 + sizeof(NX_PACKET)) * 128)` (initially 32 packets instead of 128):
```
nx_packet_pool_create(&nx_pool, "nx_pool", 1536, nx_ip_pool, sizeof(nx_ip_pool))))
nx_ip_create(&nx_ip, "nx_ip", IP_ADDRESS(0, 0, 0, 0), IP_ADDRESS(255, 255, 255, 0), &nx_pool, nx_driver_handler, nx_ip_stack, sizeof(nx_ip_stack), CONFIG_NX_HELPER_IP_PRIORITY)))
nx_arp_enable(&nx_ip, (void *)nx_arp_cache, sizeof(nx_arp_cache))
nx_tcp_enable(&nx_ip)
nx_ip_fragment_enable(&nx_ip)
...
```

The HTTPS client is created with TLS setup callback to setup the session, configure ciphers, etc. The RootCA of the server is added (working as expected), the client metadata size is 20000 bytes and the TLS packet buffer is 18500 bytes. The HTTPS client thread priority is 5. It is a lot inspired of https://github.com/TiejunMS/netxduo_projects/blob/main/issue_107/main.c.

Then the client is used to perform requests. As I indicated in the introduction, small requests (<1kB) are performed properly. Now comes downloading the firmware update binary file, this is a GET request. The function `nx_web_http_client_response_body_get` is called with `5 * NX_IP_PERIODIC_RATE` timeout.
Debugging and adding some logs I have observed the following:
- The nx-driver is getting data from the Wi-Fi module and 128 packets (so the full nx_ip_pool!!) are transmitted to NetX layer using `_nx_tcp_socket_driver_packet_receive` function;
- `NX_IP_RECEIVE_EVENT` event in NetX is generated after the first packet, but despite the event is generated, dispatching of TCP packets occurs only once the 128 packets have been pushed to NetX => **This is the first point I don't understand, as the priority of the nx-driver thread is lower that the priority of nx_ip.**
- So then NetX processes the 128 packets (128x `_nx_ip_dispatch_process` calls occur);
- The HTTP client thread is resumed after the first packet is processed (https://github.com/eclipse-threadx/netxduo/blob/v6.4.1_rel/common/src/nx_tcp_socket_state_data_check.c#L1161 is called), but despite the HTTP client thread is resumed, it doesn't process the packet and particularly the function `nx_web_http_client_response_body_get` doesn't exit; => **This is the main problem I don't understand.**
- So the 128 packets are processed in NetX but then .... nothing;
- The `nx_web_http_client_response_body_get` function returns once the `5 * NX_IP_PERIODIC_RATE` timeout elapsed, with error code 1 = `NX_NO_PACKET`;
- Then the function `nx_web_http_client_response_body_get` is called again and returns with error code 0x103 = `NX_SECURE_TLS_INVALID_STATE`.

I also tried to call the function `nx_web_http_client_response_body_get` with `NX_WAIT_FOREVER` timeout. It never returns :-(

I have seen similar issues on Github or other places, indicating the nx_pool size should be increased. It's already 128 packets so it's very large, I cannot increase again, it is already consuming a lot of memory in the MCU.

**So, WHY is the `nx_web_http_client_response_body_get` not returning the data ????**
It seems the documentation of NetX doesn't help on how to choose NetX settings (pool size, priorities etc) in a larger application with other threads etc. I haven't found troubleshooting guide either so I'm trying to debug ... not very easy.

Thanks a lot for any ideas there!
Joel

fdesbiens, 2025-04-04:
Thank you for reaching out, @joelguittet .

@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: please provide feedback.

joelguittet, 2025-04-05:
@fdesbiens thanks. Just for information: bumped to threadx 6.4.2 and netxduo 6.4.3 (latest releases available today), issue is exactly the same.

rahmanih, 2025-04-15:
> [@fdesbiens](https://github.com/fdesbiens) thanks. Just for information: bumped to threadx 6.4.2 and netxduo 6.4.3 (latest releases available today), issue is exactly the same.

Hi @joelguittet,
the issue is most likely in the WiFi module Firmware. In Fact the **B-L4S5I-IOT01A**, embeds an Inventek es_wifi module that has its own TCP/IP stack. Could you try to update the es_wifi Firmware and retest.

If the problem persists, you may fill the issue under https://community.st.com

regards
Haithem.

joelguittet, 2025-04-22:
Hello @rahmanih 

I know the wifi firmware but I think your response is a bit simple...
Particularly I have written:

> The nx-driver is getting data from the Wi-Fi module and 128 packets (so the full nx_ip_pool!!) are transmitted to NetX layer using _nx_tcp_socket_driver_packet_receive function; The 128 packets represent more than 100kB of data

I have made debugging sessions to understand the issue and tried to post here the most accurate understanding I can. I see the packets from the wifi module being recorded in the nx pool until it is full. This shows the wifi module is working as expected, but netxduo doesn't. I let you check again the details of my first message here to see why I think that.

As indicated in the first message, communication is also working fine with small messages, now come a big HTTPS GET to do firmware update. So again conforts the module is working fine, except you point me a specific bug from Inventek that is specific to this scenario.

Thanks for the support,
Joel

Ps : I will check for the wifi firmware version asap and post here.

joelguittet, 2025-04-29:
Hello @rahmanih 

For information my Wi-Fi module has the latest available version : 3.5.2.7.

Joel

joelguittet, 2025-05-05:
Hello @rahmanih @fdesbiens 

No news on this topic, totally blocked actually... I'm available for deeper investigation/logs/etc if required and really need help to understand the issue.

Thanks for the support,
Joel

joelguittet, 2025-05-06:
Hello @rahmanih 

Why do you think this could be a reference? I don't see anything relevant to my issue there. Can you point specific things in their source code maybe ?

I do HTTPS, it is MQTT.
I do firmware upgrade, so large payload, they do telemetry only.

Thanks
Joel

rahmanih, 2025-05-06:
Hi @joelguittet,
Isn't it possible to use this project to connect to your server and download the binary file, to test whether it works or not?
another suggestion could be to use the B-U585-IOT board that embeds a mx_chip wifi module that supports the "bypass" mode.
i.e all the TCP/IP management is under NetXDuo stack.

joelguittet, 2025-05-06:
@rahmanih no, changing the server is absolutely not an option. I should check other boards, I have an stm32h7 with Ethernet interface to test with, but purpose was to do WiFi...

@fdesbiens is it possible to escalade this issue, so that somebody can analyse the debugging I have done and written in the initial message and provide a feedback? Thanks!

StuartGorman, 2025-05-30:
I'm not sure if my experience is related to yours, but I have experienced a similar issue with HTTPS downloads. The issue I experienced was that the HTTPS client decodes the TLS frame and tries to inject it back into the packet pool. If the pool doesn't have enough space if will fail and then you won't get your data but it is not obvious what happened. To resolve, I set a secondary pool that is large enough to fit a full TLS frame (16kB) with `nx_secure_tls_session_packet_pool_set`. You may also need to pass the secondary pool into `nx_web_http_client_create`.

joelguittet, 2025-06-12:
Thanks for the suggestion @StuartGorman. My pool is quite large actually so didn't except to be an issue to be honnest.

@fdesbiens @rahmanih I finaly managed to get it working! Was not very easy, no bug in NetxDuo as I was almost sure of that, but configuration modified a bit. I have done severals so not sure which one effectily solves the issue.

The problem is also related to an other issue I have already opened long time ago, and related to the HTTP client skeleton we need to implement to perform the GET operation: https://github.com/eclipse-threadx/netxduo/issues/261
No clear example found and moreover the documentation is wrong: https://github.com/eclipse-threadx/rtos-docs/blob/main/rtos-docs/netx-duo/netx-duo-web-http/chapter3.md#nx_web_http_client_response_body_get => this function at least also returns `NX_NO_PACKET` (not an error just indicating no packet received before the end of the timeout, and `NX_NOT_CONNECTED` (which indicates the remote server closed the connection at the end of the download).
Those 2 values must be considered when using `nx_web_http_client_response_body_get` function.

Hope this can help people reading this issue...