eclipse-threadx/netxduo #314: [PPP] NetX Duo: PPP link established, but no IP traffic (UDP/TCP/ICMP fail)

fransysget, 2025-04-27:
We are trying NetX Duo and the PPP add-on to establish a data session using a SIMCom A7683E module UART connected to STM32H7 microcontroller. We receive data using RX DMA by calling `HAL_UARTEx_ReceiveToIdle_DMA(...)` and from the callback `HAL_UARTEx_RxEventCallback(...)` we send the received buffer to the AT commands parser or to a PPP feeder thread depending the modem status. 

At this point, we should say that PPP negotiation completes successfully (`nx_ppp_status_get(...)` returns `NX_PPP_STATUS_ESTABLISHED` and `nx_ip_status_check()` returns `NX_IP_LINK_ENABLED`), so we obtain a valid IP address when we call `nx_ip_address_get(...)`. 

However, even though the link is established:

1.  if we send UDP datagrams nothing is arriving to destination... 
2. Also, if we try to establish a TCP connection then `nx_tcp_client_socket_connect()` returns `NX_NOT_CONNECTED` error. I can see PPP TCP SYN packets are correctly formed and transmitted repeatedly, but no SYN-ACK is ever received by NetX Duo. 
3. Also, we enable ICMP in NetXDuo but any PING sent from remote server (which has access to modem assigned IP 10.x.x.x) fails with 0 packets received, all lost. 

Also comment that we've done previous test with AT commands to send UDP datagrams, send/receive data with TCP and everything works fine. So, it is not a problem of the simcard, network status, remote server destination, firewall, etc. 
 
Let me explain carefully steps we have followed and that we believe that everything has been done as specified in github documents. 

1) Initialize modem
We have sent the following commands:
AT
AT+CFUN?                           (Answer: +CFUN: 1)
AT+CSCLK?                          (Answer: +CSCLK: 0)
AT+CPIN?                           (Answer: +CPIN: READY)
AT+CREG?                           (Answer: +CREG: 1,5)
AT+CGREG?                          (Answer: +CGREG: 1,5)
AT+CEREG?                          (Answer: +CEREG: 1,5)
AT+CSQ?                            (Answer: +CSQ: 27,99)
AT+COPS?                           (Answer: +COPS: 0,2,"21407",7)
AT+CGAUTH=1,1,"my_user","my_pwd"
AT+CGDCONT=1,"IP","myapn.com"

2) Dial Internet Service Provider (ISP) & Wait for Connection
ATD*99#       (Answer: CONNECT 115200)

3) Start NetX Duo PPP
Here is our NetXDuo initialization code... We have removed in here ThreadX and NetX error control blocks in order to simplify presentation, but in real code we check errors and no error occurs during initialization

```c
#define NX_MEM_POOL_SIZE        (32768)
#define PACKET_PAYLOAD_SIZE       (512)
#define PACKET_POL_SIZE         (16384)
#define PPP_THREAD_STACK_SIZE    (4096)
#define PPP_THREAD_PRIORITY         (10)
#define IP_THREAD_STACK_SIZE     (4096)
#define IP_THREAD_PRIORITY       (PPP_THREAD_PRIORITY + 1)

UINT ppp_driver_init()
{
    UINT status;
    void* memory_ptr;

    /* prevent more than one initialization... */
    if (_nx_initialized)
    {
        return NX_SUCCESS;
    }

    status = tx_byte_pool_create( &nx_byte_pool, "NX Memory pool", nx_byte_pool_buffer, NX_MEM_POOL_SIZE);

    /* Initialize the NetX system */
    nx_system_initialize();

    /* Packet pool memory reservation */
    status = tx_byte_allocate(&nx_byte_pool, &memory_ptr, PACKET_POL_SIZE, TX_NO_WAIT);

    /* NetXDuo Packet Pool creation */
    status = nx_packet_pool_create(&packet_pool,
                                   "NX Packet Pool",
                                   PACKET_PAYLOAD_SIZE,
                                   memory_ptr,
                                   PACKET_POL_SIZE);

    /* Allocating memory for the PPP thread stack */
[... 241 lines cut ...]
    // We are not arriving to this point because of NX_NOT_CONNECTED error in nx_tcp_client_socket_connect()...

}
```

And we log the following PPP frame sent to the modem: 

```
PPP Frame #18 [TX] at 58187 ms (85 bytes)
7E FF 7D 23 7D 20 21 45 7D 20 7D 20 30 7D 20 7D 23 7D 20 7D 20 80 7D 26 73 
F7 7D 2A 72 C0 4B 7D 37 C0 E4 50 D7 7D 30 7D 20 50 B3 E3 D9 5F 7D 20 7D 20
7D 20 7D 20 70 7D 22 7D 24 7D 20 56 AF 7D 20 7D 20 7D 22 7D 24 7D 25 B4 7D 
21 7D 21 7D 21 7D 20 AB 25 7E    
```

It seems to me a perfectly well formed IP packet with TCP SYN. But no SYN-ACK is received back so it ends with error NX_NOT_CONNECTED.

Is there something we are doing wrong?
Is there something obvious we are missing? We have verified modem configuration, SIM, network, server accessibility, and PPP link establishment, but no IP traffic seems to be received.

We do not know what else to try/check, etc. 

Any help would be greatly appreciated

Best regards,

fdesbiens, 2025-04-29:
@rahmanih: can you help, please?

EdouardMALOT, 2025-04-29:
Hi @fdesbiens,

I use PPP on netxduo.

Below my initialization, I hope it can helps :

`/**
 * ******************************************************************************
 *  Function Name: init_4G()
 * ******************************************************************************
 *
 */
void init_lte(void)
{
    // Initialize the NetXDuo system.
    nx_system_initialize();

    // Ip pool
    if (nx_packet_pool_create(&NxAppPool, "NetXDuo Pool", DEFAULT_PAYLOAD_SIZE, &packet_pool[0], NX_APP_PACKET_POOL_SIZE) != NX_SUCCESS)
    {
        Error_Handler();
    }

    //Create GSM instance
    init_gsm_instance(&GsmIpInstance, &NxAppPool);

	//Enable fragmentation
	if (nx_ip_fragment_enable(&GsmIpInstance) != NX_SUCCESS)
	{
		Error_Handler();
	}

	// Enable ICMP (Ping)
	if (nx_icmp_enable(&GsmIpInstance) != NX_SUCCESS)
	{
		Error_Handler();
	}

	//TCP Enable
	if (nx_tcp_enable(&GsmIpInstance) != NX_SUCCESS)
	{
		Error_Handler();
	}

	//UDP Enable
	if (nx_udp_enable(&GsmIpInstance) != NX_SUCCESS)
	{
		Error_Handler();
	}

    // TLS module
    #ifdef NX_SECURE_ENABLE
        nx_secure_tls_initialize();
    #endif

	init_dns_mutex();
	init_dns(&GsmIpInstance);
	dns_add_server_ipv4(GOOGLE_DNS_IP);

	launch_mqtt_client_task(&GsmIpInstance);
}`

`/**
 * ******************************************************************************
 *  Function Name: init_gsm_instance()
 * ******************************************************************************
 *
 */
void init_gsm_instance(NX_IP* pGsmIpInstance, NX_PACKET_POOL* pPacket_Pool)
{
	//Start Uart Rx
	HAL_UART_DeInit(&UART_GSM);
	HAL_UART_Init( &UART_GSM);

	//Init Semaphoe
	uart_tx_dma_end = osSemaphoreNew (1, 0, NULL);
	if(uart_tx_dma_end == NULL)
	{
	   Error_Handler();
	}

	GSM_StartRx();

	//PPP
	uint8_t *pointer = MemAlloc(Nx_PPP_INSTANCE_THREAD_SIZE, RTOS2_BYTE_POOL_STACK_TYPE);
	if(pointer == NULL)
	{
		Error_Handler();
	}
	if (NX_SUCCESS != nx_ppp_create(&ppp_gsm, "PPP", pGsmIpInstance, pointer, Nx_PPP_INSTANCE_THREAD_SIZE, (osPriorityISR - NX_PPP_INSTANCE_PRIORITY), pPacket_Pool, ppp_invalid_packet_handler, ppp_serial_byte_output, ppp_serial_byte_done))
	{
		Error_Handler();
	}

	/* Register the link up/down callbacks.  */
	if (NX_SUCCESS != nx_ppp_link_up_notify(&ppp_gsm, ppp_link_up_callback))
	{
		Error_Handler();
	}

	if (NX_SUCCESS != nx_ppp_link_down_notify(&ppp_gsm, ppp_link_down_callback))
	{
		Error_Handler();
	}

	// Create GSM IP instance
	pointer = MemAlloc(Nx_IP_INSTANCE_THREAD_SIZE, RTOS2_BYTE_POOL_STACK_TYPE);
	if(pointer == NULL)
	{
		Error_Handler();
	}

	if (NX_SUCCESS != nx_ip_create(pGsmIpInstance, "GSM Ip instance",  0, 0, pPacket_Pool, nx_ppp_driver, (CHAR *)pointer, Nx_IP_INSTANCE_THREAD_SIZE, (osPriorityISR - NX_IP_INSTANCE_PRIORITY)))
	{
		Error_Handler();
	}
}`

Then I wait until a got an IP : 

`
		  //Check if IP change
		  //------------------
			  ULONG IpAddress = 0;
			  ULONG NetMask = 0;
			  ULONG NetGw = 0;

			  if (nx_ip_address_get(&GsmIpInstance, &IpAddress, &NetMask) == TX_SUCCESS)
			  {
				  if(IpAddress != gsm_config.ip || NetMask != gsm_config.mask)
				  {
					  //Print IP
					  //-------
						  log_info("\tGSM Ip   : %lu.%lu.%lu.%lu",  (IpAddress >> 24) & 0xff,
																	(IpAddress >> 16) & 0xff,
																	(IpAddress >> 8) & 0xff,
																	(IpAddress& 0xff) );

					  //Print Mask
					  //----------
						  log_info("\tGSM Mask : %lu.%lu.%lu.%lu",  (NetMask >> 24) & 0xff,
																	(NetMask >> 16) & 0xff,
																	(NetMask >> 8) & 0xff,
																	(NetMask& 0xff) );

					  //Print GW
					  //--------
						  nx_ip_gateway_address_get(&GsmIpInstance, &NetGw);
						  log_info("\tGSM Gw   : %lu.%lu.%lu.%lu",  (NetGw >> 24) & 0xff,
																	(NetGw >> 16) & 0xff,
																	(NetGw >> 8) & 0xff,
																	(NetGw& 0xff) );
					  //Start/Stop LED + mDNS
					  //---------------------
						  if(IpAddress != 0)
						  {
						  	mqtt_start();
							  //led_gsm(true);
						  }else{
							  //led_gsm(false);
							delete_udp_to_serveur();
						  }

					  //Update value
						  gsm_config.ip   = IpAddress;
						  gsm_config.mask = NetMask;
						  gsm_config.gw   = NetGw;
				  }
			  }
`

fransysget, 2025-04-30:
Hi @EdouardMALOT,

Thanks for sharing your initialization routine — I really don’t see any significant differences compared to ours.

One thing I noticed is that your `nx_ppp_create()` call includes an additional parameter `ppp_serial_byte_done`, which I haven’t seen documented. I'd appreciate it if you could point me to any reference on that, just in case it's relevant.

Regarding the NetX Duo version, I'm using **IAR IDE v9.20**, and the `_nx_version_id[]` string reports:  
`"Copyright (c) Microsoft Corporation. All rights reserved.  *  NetX Duo Cortex-M7/IAR Version 6.4.0 *"`.

I also noticed your `nx_ip_create()` call uses `0` for the `network_mask` parameter. I've updated ours to match that — just in case it makes a difference — but unfortunately, the issue persists.

So at this point, the situation remains unchanged. If you have any suggestions or tests I could run to further diagnose what's going wrong, I’d be more than happy to try them out.

Again, any help is greatly appreciated

Thanks again for your support!

EdouardMALOT, 2025-04-30:
@fransysget

Yes, I’ve made a few updates to the PPP files:

- ppp_serial_byte_output: This function originally sends data byte by byte, which isn’t ideal for DMA transfers. I now use ppp_serial_byte_output() to store the bytes in a buffer, and I’ve modified PPP to call ppp_serial_byte_done() once a complete PPP frame is ready. At that point, I transfer the entire buffer using DMA.

- I'm using NetX Duo version 6.2.

Regarding modules, I’ve tested SIM7672, SIM800, and UBLOX LARA-R6 without any issues.

Below AT commands to initialise SIM7672G : 

[      0][INF] GSM> LTE Power ON
[      0][DBG] GSM> PWR KEY = LOW
[      0][DBG] GSM> Power ON
[      0][DBG] GSM> GSM_STATE : 0 to 4
[      0][WRN] GSM> ppp_link_down : PPP_STOP
[      0][INF] GSM> LTE Power ON
[      0][DBG] GSM> PWR KEY = LOW
[      0][DBG] GSM> Power ON
[      0][DBG] GSM> GSM_STATE : 0 to 4
[      0][DBG] GSM> PWR KEY = HIGH
[      0][DBG] GSM> GSM_STATE : 4 to 5
[      0][DBG] GSM> Check if GSM starts
[      0][DBG] GSM>  -> ATE0  
[      0][DBG] GSM>  <-   QCRDY  ATE0   OK  
[      1][DBG] GSM>  -> ATE0  
[      1][DBG] GSM>  <-   OK  
[      1][DBG] GSM>  -> AT+CNBP?  
[      1][DBG] GSM>  <-   +CNBP: 0X000001E20B0E18DF, 0X02    OK  
[      1][INF] GSM>   +CNBP: 0X000001E20B0E18DF, 0X02    OK  
[      1][DBG] GSM>  -> AT+CGMR  
[      1][DBG] GSM>  <-   +CGMR: 2368B01SIM767XM5A_M    OK  
[      1][INF] GSM>   +CGMR: 2368B01SIM767XM5A_M    OK  
[      1][DBG] GSM>  -> AT+CMEE=2  
[      1][DBG] GSM>  <-   +CPIN: READY    OK    SMS DONE  
[      1][DBG] GSM>  -> AT+CMEE=2  
[      1][DBG] GSM>  <-   OK  
[      1][DBG] GSM>  -> AT+CICCID  
[      1][DBG] GSM>  <-   +ICCID: 8944477200002189604    OK  
[      1][INF] GSM>   +ICCID: 8944477200002189604    OK  
[      1][DBG] GSM>  -> AT+IFC=2,2  
[      1][DBG] GSM>  <-   OK  
[      1][DBG] GSM>  -> AT+IFC?  
[      1][DBG] GSM>  <-   +IFC: 2,2    OK  
[      1][DBG] GSM>  -> AT+IPR=921600  
[      1][DBG] GSM>  <-   OK  
[      1][DBG] GSM>  -> AT+IPR?  
[      1][DBG] GSM>  <-   +IPR: 921600    OK  
[      2][DBG] GSM>  -> AT+CPIN?  
[      2][DBG] GSM>  <-   +CPIN: READY    OK  
[      2][DBG] GSM>  -> AT+CFUN=4  
[      2][DBG] GSM>  <-   OK  
[      2][DBG] GSM>  -> AT+CGDCONT=1,"IP","",,0,0  
[      2][DBG] GSM>  <-   OK  
[      2][DBG] GSM>  -> AT+CFUN=1  
[      2][DBG] GSM>  <-   OK  
[      2][DBG] GSM>  -> AT+CGATT?  
[      3][DBG] GSM>  <-   +CGATT: 0    OK  
[      3][DBG] GSM>  -> AT+CGATT?  
[      4][DBG] GSM>  <-   +CGATT: 0    OK  
[      4][DBG] GSM>  -> AT+CGATT?  
[      5][DBG] GSM>  <-   +CGATT: 0    OK  
[      5][DBG] GSM>  -> AT+CGATT?  
[      6][DBG] GSM>  <-   +CGATT: 0    OK  
[      6][DBG] GSM>  -> AT+CGATT?  
[      6][DBG] GSM>  <-   +CGATT: 1    OK  
[      6][DBG] GSM>  -> AT+CREG?  
[      7][DBG] GSM>  <-   +CREG: 0,6    OK  
[      7][INF] GSM>   +CREG: 0,6    OK  
[      7][DBG] GSM>  -> AT+COPS?  
[      8][DBG] GSM>  <-   +COPS: 0,2,"20810",7    OK  
[      8][INF] GSM>   +COPS: 0,2,"20810",7    OK  
[      8][DBG] GSM>  -> AT+CGACT=1,1  
[      8][DBG] GSM>  <-   OK  
[      8][DBG] GSM>  -> AT+CGCONTRDP=1  
[      9][DBG] GSM>  <-   +CGCONTRDP: 1,5,"iot.truphone.com","10.189.213.5",,"10.31.121.84","10.31.121.85",,,,,1430    OK  
[      9][INF] GSM>   +CGCONTRDP: 1,5,"iot.truphone.com","10.189.213.5",,"10.31.121.84","10.31.121.85",,,,,1430    OK  
[      9][DBG] GSM>  -> AT+CSQ  
[     10][DBG] GSM>  <-   +CSQ: 17,0    OK  
[     10][INF] GSM>   +CSQ: 17,0    OK  
[     10][DBG] GSM> GSM Start PPP
[     10][DBG] GSM>  -> ATD*99#  
[     10][DBG] GSM>  <-   CONNECT  
[     10][INF] GSM> PPP is active
[     10][INF] GSM> link_up_callback
[     10][INF] LTE> 	GSM Ip   : 10.189.213.5
[     10][INF] LTE> 	GSM Mask : 255.255.255.255
[     10][INF] LTE> 	GSM Gw   : 10.189.213.5

Note: I also noticed that you're calling nx_ppp_ip_address_assign(). I’ve never used that function myself.  
Did you try removing it ?

fransysget, 2025-05-01:
Hi @EdouardMALOT,

Thanks again for your detailed reply and for pointing out your `ppp_serial_byte_output` and `ppp_serial_byte_done` optimization. It definitely looks interesting, although at this stage I'm still focused on just getting the basic PPP data path working before diving into performance improvements. Still, good to know for when I get past this first hurdle!

As suggested, I removed the call to `nx_ppp_ip_address_assign()` from my NetX Duo PPP initialization routine, but unfortunately the issue remains exactly the same — no improvement so far.

I also noticed that your initialization sequence includes some AT commands that I hadn't been using before. I tested adding them to the flow on my SIMCom A7683E module, but sadly, that didn't make any difference either. However, for completeness, here are the responses I get from the modem when issuing those commands before `ATD*99#`:

```
AT+CNBP?
+CNBP: 0X0000000000000000,0X00000000080800D5,0X0000

OK

AT+CGATT?
+CGATT: 1

OK

AT+CGACT=1,1

OK

AT+CGCONTRDP=1
+CGCONTRDP: 1,5,"wlapn.com.mnc002.mcc238.gprs","10.114.192.75","","10.4.0.240","10.4.0.230","","",0,01500

OK
```

I'd also like to add some PPP frame logs to show that the TCP SYN retry is occurring every 1000 ms, but unfortunately, no reply is ever received from the A7683E (no TCP SYN-ACK):

```
11:51:41 [TRCE][PPP THREAD] PPP Frame #18 [TX] at 63387 ms (85 bytes)
7E FF 7D 23 7D 20 21 45 7D 20 7D 20 30 7D 20 7D 23 7D 20 7D 20 80 7D 26 73
F7 7D 2A 72 C0 4B 7D 37 C0 E4 50 D7 7D 30 7D 20 50 B3 E3 D9 5F 7D 20 7D 20
7D 20 7D 20 70 7D 22 7D 24 7D 20 56 AF 7D 20 7D 20 7D 22 7D 24 7D 25 B4 7D
21 7D 21 7D 21 7D 20 AB 25 7E

11:51:42 [TRCE][PPP THREAD] PPP Frame #19 [TX] at 64387 ms (85 bytes)
7E FF 7D 23 7D 20 21 45 7D 20 7D 20 30 7D 20 7D 24 7D 20 7D 20 80 7D 26 73
F6 7D 2A 72 C0 4B 7D 37 C0 E4 50 D7 7D 30 7D 20 50 B3 E3 D9 5F 7D 20 7D 20
7D 20 7D 20 70 7D 22 7D 24 7D 20 56 AF 7D 20 7D 20 7D 22 7D 24 7D 25 B4 7D
21 7D 21 7D 21 7D 20 B9 98 7E

... more PPP frame of TCP SYN retries in here ... 

11:51:49 [TRCE][PPP THREAD] PPP Frame #26 [TX] at 71387 ms (85 bytes)
7E FF 7D 23 7D 20 21 45 7D 20 7D 20 30 7D 20 7D 2B 7D 20 7D 20 80 7D 26 73
EF 7D 2A 72 C0 4B 7D 37 C0 E4 50 D7 7D 30 7D 20 50 B3 E3 D9 5F 7D 20 7D 20
7D 20 7D 20 70 7D 22 7D 24 7D 20 56 AF 7D 20 7D 20 7D 22 7D 24 7D 25 B4 7D
21 7D 21 7D 21 7D 20 5E 44 7E

11:51:50 [TRCE][PPP THREAD] PPP Frame #27 [TX] at 72387 ms (85 bytes)
7E FF 7D 23 7D 20 21 45 7D 20 7D 20 30 7D 20 7D 2C 7D 20 7D 20 80 7D 26 73
EE 7D 2A 72 C0 4B 7D 37 C0 E4 50 D7 7D 30 7D 20 50 B3 E3 D9 5F 7D 20 7D 20
7D 20 7D 20 70 7D 22 7D 24 7D 20 56 AF 7D 20 7D 20 7D 22 7D 24 7D 25 B4 7D
21 7D 21 7D 21 7D 20 4C F9 7E
```

After around 10 retries, the call to `nx_tcp_client_socket_connect()` returns error `NX_NOT_CONNECTED`.

So I’m still stuck in the same situation: the PPP link is successfully established, and I get a valid IP address, but TCP connection attempts fail (no IP traffic appears to be received after sending TCP SYN), and also UDP datagrams never reach the destination either. I’m probably missing something, but I haven’t been able to pinpoint what.

If you can think of any further idea or test to try, I’d be more than happy to follow up.

Thanks again for your support!

EdouardMALOT, 2025-05-01:
@fransysget 

- when you say "no reply is ever received from the A7683E" you mean the module doesn't send any reply on UART ?

- CGCONTRDP shows the modem MRU is 1500.
     Is NX_PPP_MRU equal to or greater than 1500 in netX?

- Did you try enabling the PPP logs?
#define NX_PPP_DEBUG_LOG_ENABLE
#define NX_PPP_DEBUG_LOG_PRINT_ENABLE
(You should see the LCP handshake.)

fransysget, 2025-05-02:
Hi @EdouardMALOT,

I can confirm that `NX_PPP_MRU` is exactly `1500`, and this value is sent in the first LCP frame along with options for Protocol Field Compression and Address and Control Field Compression.

As requested, here's the log of the PPP frames (with a slightly reformatted layout for clarity):

```
Time: 55352, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0; Authenticated
[TX] (14 bytes): C0 21 01 15 00 0C 01 04 05 DC 07 02 08 02

Time: 55398, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated
[RX] (26 bytes): C0 21 01 01 00 18 02 06 00 00 00 00 03 04 C0 23 05 06 4F 6A 96 6E 07 02 08 02

Time: 55431, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated
[TX] (10 bytes): C0 21 04 01 00 08 03 04 C0 23

Time: 55460, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated
[RX] (14 bytes): C0 21 02 15 00 0C 01 04 05 DC 07 02 08 02

Time: 55491, PPP State: 1, PPP LCP State: 3, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated
[RX] (22 bytes): C0 21 01 02 00 14 02 06 00 00 00 00 05 06 4B 9C 24 57 07 02 08 02

Time: 55518, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated
[TX] (22 bytes): C0 21 02 02 00 14 02 06 00 00 00 00 05 06 4B 9C 24 57 07 02 08 02

Time: 55552, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 1, Authenticated
[TX] (12 bytes): 80 21 01 16 00 0A 03 06 00 00 00 00

Time: 55597, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated
[RX] (6 bytes):  80 21 01 03 00 04

Time: 55619, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 2, Authenticated
[TX] (6 bytes): 80 21 02 03 00 04

Time: 55646, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 4, Authenticated
[RX] (24 bytes): 80 21 03 16 00 16 03 06 0A 72 C0 4B 81 06 0A 04 00 F0 83 06 0A 04 00 E6

Time: 55674, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 4, Authenticated
[TX] (12 bytes): 80 21 01 17 00 0A 03 06 0A 72 C0 4B

Time: 55714, PPP State: 1, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 4, Authenticated
[RX] (12 bytes): 80 21 02 17 00 0A 03 06 0A 72 C0 4B

✅ At this point, PPP link is up (`NX_PPP_STATUS_ESTABLISHED`), and IP address `10.114.192.75` has been assigned (with mask `255.255.255.255`, and gateway same as IP).

✅ About to send UDP message with "Hello\r\n" to 10.4.17.62:6970! ----------------------------

Time: 57394, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (38 bytes): 00 21 45 00 00 24 00 01 00 00 80 11 54 C9 0A 72 C0 4B 0A 04 11 3E CD EE 1B 3A 00 10 02 C7 48 65 6C 6C 6F 0D 0A 00

✅ The frame seems correct and complete (IP + UDP headers and payload verified), but **nothing is received on the destination server**.

✅  TCP test connection (------------------------------------------------------------------------

Time: 62406, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 02 00 00 80 06 73 F8 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 63314, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 03 00 00 80 06 73 F7 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 64314, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 04 00 00 80 06 73 F6 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 65314, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 05 00 00 80 06 73 F5 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

[... 13 lines cut ...]
[TX] (50 bytes): 00 21 45 00 00 30 00 09 00 00 80 06 73 F1 0A 72 C0 4B 17 C0 E4 50 D7 10 00
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 70314, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 0A 00 00 80 06 73 F0 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 71314, PPP State: 2, PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 0B 00 00 80 06 73 EF 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
                 50 B3 E3 D9 5F 00 00 00 00 70 02 04 00 56 AF 00 00 02 04 05 B4 01 01 01 00

Time: 72314; PPP State: 2; PPP LCP State: 8, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 8, Authenticated
[TX] (50 bytes): 00 21 45 00 00 30 00 0C 00 00 80 06 73 EE 0A 72 C0 4B 17 C0 E4 50 D7 10 00 
[... cut ...]
✅  Each TCP SYN packet is properly formatted and sent every ~1 second. After ~10 attempts `nx_tcp_client_socket_connect()` returns `NX_NOT_CONNECTED`.

```
### Final thoughts 

The PPP negotiation looks fine but it seems that the SIMCom A7683E is either **not forwarding the traffic to the mobile network**, or it’s **blocking responses from reaching the MCU** over UART.

If you have further suggestions or ideas to help pinpoint the issue, I would really appreciate it.

Thanks again for your time and help!

fransysget, 2025-05-07:
Hi everyone, 

We're still stsuck: PPP is up, IP assigned, but no working IP traffic. 

Any ideas or experiences with similar symptoms?
- Is there a chance the SIMCom A7683E is silently filtering/ignoring packets after PPP negotiation?
- Could there be any requirement on gateway handling or missing AT configuration?
- a problem with the SIM provider? Is it possible that it works with AT commands but it doesn't with PPP?
- what else could prevent NetX Duo from receiving replies?

Thanks for your time and support!

Hnz2, 2025-05-09:
Hi @fransysget 

Personally I would blame Simcom module. At one application we use A7670E and we saw many issues with this module, regardless we not use ppp mode. LTE is generally much more problematic in comparison with good old 2G/3G.

My recommendation. If you are able, test with different module.

Jan

fransysget, 2025-05-09:
Hi @Hnz2

Thanks for your input. However, I'm confident that the issue is not related to the SIMCom module itself. Other companies I know are using the same module with PPP successfully. In fact, in this same thread, @EdouardMALOT mentioned that he's been using other modules from SIMCOM (SIM7672 and SIM800) with PPP without issues. 

I've also tested with SIMCom A7672E and A7682E modules and also with a different SIM card provider, but unfortunately the results are the same. So I guess it is not related with sim card provider.

Please, @rahmanih, @fdesbiens - as I mentioned earlier, time is running out and I'm really stuck. I'm almost certain this is due to something I'm doing wrong in my use of NetX Duo, nor the hardware. If either of you (or anyone else) could spare a few minutes to review my code (posted above) and help me find the issue or suggest a new direction to explore, I would really appreciate it. 

Again, thank you for your time and support

Best regards,
Fran

fdesbiens, 2025-05-09:
Hi @fransysget.

First of all, I understand where you are coming from. You are under pressure and need a solution as soon as possible. However, Eclipse ThreadX is an open source project. Any support we can provide is on a best-effort basis. Our committers all have day jobs, which is their first priority.

There is no way I can "escalate" your issue. We are a very small team, and some of us do not have expertise with the entire stack. Take me, for example: I do not have access to hardware similar to yours and know nothing about PPP. Moreover, I have no authority over the committers. I can nudge them but, ultimately, they will decide what tasks they pick up on their own. 

@rahmanih already took some time to look at your issue. Maybe he will have further insights next week. May 8 is a holiday in his country and I suspect he took today off as well. 

If your project is critical and time-sensitive, then we can introduce you to one of our ThreadX commercial support partners. Please email threadx-info@eclipse.org and we will point you to a company with the required skills.

fransysget, 2025-05-09:
Hi @fdesbiens,

Thank you very much for your thoughtful and transparent reply — I truly appreciate it.

I fully understand and respect the nature of open source projects and the fact that your time and support are provided on a best-effort basis, often on top of your regular responsibilities. I didn’t mean to put pressure or come across as demanding — I’m just at a point where I’ve exhausted all the ideas and debugging avenues I could think of, and I was sincerely hoping that maybe someone with more familiarity with the stack could spot something I’m missing.

Please know that my intention was never to impose pressure, and I truly hope no one felt uncomfortable because of my messages. If that was the case, I sincerely and humbly apologize.

Also, thank you for offering an alternative via commercial support. That’s a great option to consider, and I will keep it in mind depending on how things evolve over the next few days.

Again, many thanks to you and the team for the time and help you’ve already provided. I’ll remain attentive in case anything else is suggested in the future.

Best regards,
Fran

EdouardMALOT, 2025-05-09:
@fransysget,
If you want I can handle your issue (this weekend or next week)

Can you contact me on LinkedIn ?

fdesbiens, 2025-05-10:
Hi @fransysget.

> Please know that my intention was never to impose pressure, and I truly hope no one felt uncomfortable because of my messages. If that was the case, I sincerely and humbly apologize.

You do not have to apologise! As I said, I understand where you are coming from. I sincerely wish I could help more. I just wanted to make sure to set clear expectations, since you seem to be on a tight timeline.

Best,

Frédéric

fransysget, 2025-05-12:
Hi,

Thanks @fdesbiens for your kind reply and for your understanding

Thanks @EdouardMALOT for your offer to help -- I'll get in touch with you via LinkedIn

But I have a follow-up question: If NetX Duo is generating the PPP frames and successfully sending them over UART to the SIMCom A7683E module — and assuming that these frames are correctly formed (which seems to be the case right now) — If PPP link is up (NX_PPP_STATUS_ESTABLISHED), and IP address is assigned... would that mean the issue is no longer related to the NetX Duo library?

In this situation, should I instead focus on investigating the A7683E module’s configuration or maybe potential limitations at the SIM card/network level (e.g., blocked services, etc.)?

Thanks again for your support and guidance.

Best regards,
Fran

fransysget, 2025-06-09:
Hi again,

After analyzing this issue further in coordination with the technical team at SIMCom, I wanted to provide an update which might help clarify the root cause from a user perspective.

The PPP connection is successfully negotiated using NetX Duo, and the LCP phase properly proposes compression options (ACFC and PFC) when the macro `NX_PPP_COMPRESSION_ENABLE` is defined, as expected.

However, we have observed that even after the compression negotiation succeeds (i.e., the peer accepts ACFC and/or PFC), the IP frames sent afterward do not actually reflect this. In particular, the transmitted PPP/IP frames do not omit the Address and Control fields (ACFC) nor apply Protocol Field Compression (PFC), even though the negotiation clearly enabled them.

As a consequence of this mismatch, the SIMCom A7683E module rejects these non-compliant PPP/IP frames, and no IP-level communication is possible (e.g., a simple PING fails).

From a user perspective, and with full humility, I would expect the behavior to follow this logic:

If `NX_PPP_COMPRESSION_ENABLE` is not defined, then neither proposing nor accepting any compression options during LCP makes sense — and that already works as intended.

If `NX_PPP_COMPRESSION_ENABLE` is defined, then the stack should be able to negotiate compression based on what the peer accepts. In this case, it might make sense to store the outcome of the LCP negotiation in dedicated field/s inside the `NX_PPP` structure — for example, flags to track whether ACFC and/or PFC are enabled. Then, when transmitting PPP/IP packets, the implementation should honor that negotiation outcome and apply the corresponding compression.

Currently, as far as we can see, the transmission path (in particular functions like `_nx_ppp_process_deferred_ip_packet_send()` and `_nx_ppp_packet_transmit()`) does not take into account whether compression was actually agreed upon in LCP negotiation — and always sends the full 0xFF 0x03 0x00 0x21 header (no ACFC or PFC applied), regardless of the negotiation result.

If this understanding is incorrect, I would greatly appreciate any clarification. Otherwise, perhaps this observation could be of use to improve the PPP implementation in future updates.

Thank you again for the great work on NetX Duo.

closed 2026-09-28 by fdesbiens