eclipse-threadx/netxduo #316: HTTP Body Cannot Be Retrieved When Headers Span Multiple Packets

yam-mas, 2025-05-15:
**Describe the bug**
_nx_web_http_client_response_body_get function returns NX_WEB_HTTP_BAD_PACKET_LENGTH.

While reviewing the internal implementation of the HTTP client in `nx_web_http_client.c`, we have observed certain behaviors in internal function which are called from `_nx_web_http_client_response_body_get`.

`_nx_web_http_client_process_header_fields` appears to parse only the first packet and does not traverse any chained packets via `nx_packet_next`. As a result, if the terminating sequence of the HTTP header (`\r\n\r\n`) is not contained within the first packet, the function returns `0`. This return value is then treated as an error in `_nx_web_http_client_response_body_get`, which responds by returning `NX_WEB_HTTP_BAD_PACKET_LENGTH` and resetting the connection state.

**To Reproduce**
These behaviors were observed when the HTTP response headers exceeded the configured packet size, resulting in the headers being fragmented across multiple packets. Under these conditions, the following issues were encountered:

**Expected behavior**
- "packet_pptr" will be the packet pointer where the body is stored.
- We will also need the packet pointer at the beginning of the HTTP reception at the same time (to release the packet).

**Impact**
Unable to receive HTTP from a specific server.

**Logs and console output**
-

**Additional context**
We noticed that `_nx_web_http_client_content_length_get` also scans only a single packet for the `"Content-Length"` header. While we are not certain how the retrieved value is used internally, we would like to ask whether this function might be affected by a similar limitation when headers are split across multiple packets.

If there are any settings, usage patterns, or recommended practices that would help avoid such situations, we would appreciate your guidance.

fdesbiens, 2025-08-05:
Hi @yam-mas.

My apologies for not acknowledging this issue sooner. I will bring it to the team's attention.

fdesbiens, 2026-09-28:
Thanks for the report and for following up. I confirmed that the client collects headers across packets but parses only the first packet. [PR #454](https://github.com/eclipse-threadx/netxduo/pull/454) fixes the parsing and body packet positioning, with regression tests for split headers. The PR is open for review.