eclipse-threadx/netxduo #321: Hardware examples

navinreddy23, 2025-05-20:
Hi,

Since, STM32 and NXP have officially stopped supporting NetxDuo, will there be an effort made to support new/existing network devices?
For e.g. will there be example integration with various vendor boards?

Best regards,
Navin

closed 2025-05-27 by fdesbiens

hwmaier, 2025-05-28:
> So, ST will continue to work on NetX Duo and USBX

@fdesbiens You probably meant _FileX_ here instead of NetX Duo. Based on above cited announcement I understand ST has no future interest in NetXDuo nor ThreadX. They only forked _FileX_ and _UsbX_ to be usable without ThreadX and instead work with FreeRTOS and lwIP in the future.

Their own fork of NetXDuo (https://github.com/STMicroelectronics/stm32-mw-netxduo) sits at version 6.4 from June 2024 without receiving any updates since (excluding some minor changes to readme and license files).

Hnz2, 2025-05-28:
@fdesbiens I have similar information from ST as [hwmaier](https://github.com/hwmaier). They do not plan to use ThreadX/NetX Duo for new devices or do implementation of new features (e.g. hardware cryptography) for current one. For a limited time they will keep updating Cube packages of ThreadX. But I don't know how long this will be.