eclipse-threadx/netxduo #322: When will it be available, and is there a roadmap for IPSec?

dortainlu, 2025-05-22:
When will it be available, and is there a roadmap for IPSec?

[image]

fdesbiens, 2025-05-27:
@dortainlu To my knowledge, IPSEC has been removed from the codebase before ThreadX has been contributed to the Eclipse Foundation, as were the CoAP / LWM2M clients and the exFAT implementation in FileX. I will double check with the team.

fdesbiens, 2025-06-05:
I discussed the state of our IPSEC support with the members of our core team. IPSEC was added to NetX Duo several years ago but was not properly maintained due to a lack of customer demand. This means the implementation became obsolete, as it did not conform to the latest IPSEC specifications. Consequently, Microsoft removed this part of the codebase before contributing ThreadX to the Eclipse Foundation.

I will ensure that the documentation and other collateral are updated to remove references to IPsec.  

If you or others are willing to work on an up-to-date IPSEC implementation, then we would gladly accept such a contribution. None of the core team members has the time or interest to work on this, to be honest. If no one steps up to provide an implementation, I would rather remove the various IPSEC hooks currently found in the codebase.