eclipse-threadx/netxduo #324: Netx Duo: TCP acknowledgement issue

sabarinathKA, 2025-05-29:
TCP server is running on a STM32 board and TCP client is running on a PC.

problem:
netx duo not sending acknowledgement for the packet received immediately even if the  NX_TCP_IMMEDIATE_ACK is defined.
here after receiving the packet its sending the acknowledgement only when we call the nx_tcp_socket_send . here i need to process the incoming packet and do some time consuming operation so since am not calling the send function immediately after receiving the packet so acknowledgment sent is delayed so the the other peer(PC) is retransmitting the packet since its not received the acknowledgment from the board and its disconnecting the socket after 5 retransmission(since am using windows). so how can i resolve this issue here.

Hnz2, 2025-05-29:
@sabarinathKA This sounds like a thread priority issue. How do you have set IP thread priority in comparison with other threads?

sabarinathKA, 2025-05-29:
here IP thread is high priority( defined  1 , all other running thread is defined 10). here why the acknowledgment is only sent once we call the nx_tcp_socket_send API 
so after receiving if am calling any block API in the same thread its not sends the acknowledgment until the send API calls.

fdesbiens, 2025-08-05:
@sabarinathKA, you did not reply to the comment @yuxinzhou5 made. 

Did the fix he proposed work for you? If not, please reply soon, or I will close this issue due to inactivity.