eclipse-threadx/netxduo #327: How to Create Multiple IP instances with multiple physical interface?

XJTUzhangxu, 2025-06-13:
I want to create multiple IP instances that will be bind to multiple physical interfaces. I can't find an example about it. Could you share some solves to it? I will very appreciate it, thanks.

relative: [Multiple DHCP](https://github.com/eclipse-threadx/netxduo/issues/54)

fdesbiens, 2025-08-05:
Hi @XJTUzhangxu. My apologies for not getting to this issue sooner.  

Is what the requester describes possible, @yuxinzhou5?

yuxinzhou5, 2025-08-05:
NetX does not support this use case today.  You are probably aware that NetX can support multiple physical interfaces with one instance.  I am curious about your use case that would require multiple NetX instances, each for one interface. Thanks!